adafruit/circup #187: Fail to remove build-in bundle (to bypass infra issue)

dglaude, 2023-10-31:
On Windows 10, with Python 3.10.11, I try to use circup to migrate to CP 9.0.

To avoid errors, I wanted to remove a bundle, but I cannot remove it.

Not sure if that is normal.

```
C:\Project>python --version
Python 3.10.11

C:\Project>circup --version
CircUp, A CircuitPython module updater. Version 1.4.0

C:\Project>circup bundle-show
Downloading latest bundles for circuitpython/CircuitPython_Org_Bundle (20231029).
py:
There was a problem downloading that platform bundle. Skipping and using existing download if available.
adafruit/Adafruit_CircuitPython_Bundle
    https://github.com/adafruit/Adafruit_CircuitPython_Bundle
    version = 20231031
adafruit/CircuitPython_Community_Bundle
    https://github.com/adafruit/CircuitPython_Community_Bundle
    version = 20231030
circuitpython/CircuitPython_Org_Bundle
    https://github.com/circuitpython/CircuitPython_Org_Bundle
    version = 0.0.3

C:\Project>circup bundle-remove circuitpython/CircuitPython_Org_Bundle
Cannot remove built-in module:
    circuitpython/CircuitPython_Org_Bundle
```

dhalbert, 2023-10-31:
Maybe we should change `Cannot remove built-in module:` to `Not allowed to remove built-in module` or `Removing built-in module not permitted`.

dglaude, 2023-10-31:
Maybe add a `circup bundle-restore` to bring back the default in case someone want to repair.
And the message can be: `You are removing a default bundle, you can user "bundle-restore" if you change your mind or that was a mistake.`