adafruit/circup #189: Clean up circup requirements; update base Python version; switch to pyproject.toml?

dhalbert, 2023-11-10:
Motivated by #143.

I do not understand the current state of Python package management well enough to decide what to do here, but:

- #143 suggests adding `setuptools` to circup's requirements; this is because Python 3.12 no longer provides `setuptools` automatically . If we switched from`setup.py` to `pyproject.toml`, would this issue go away?
- circup has an `install_requires` in `setup.py`, and also has a `requirements.txt`. My impression is that the `requirements.txt` does not actually serve any purpose and could be deleted. Is that right?
- We are currently supporting Python 3.7 and up. Should we move forward from that? If so, to what version?

rsbohn, 2023-11-13:
>> circup install adafruit_lsm6ds
Traceback (most recent call last):
  File "<frozen runpy>", line 198, in _run_module_as_main
  File "<frozen runpy>", line 88, in _run_code
  File "C:\tools\blinka\scripts\circup.exe\__main__.py", line 4, in <module>
  File "C:\tools\blinka\Lib\site-packages\circup\__init__.py", line 25, in <module>
    import toml
ModuleNotFoundError: No module named 'toml'

[after]
Successfully installed circup-1.5.0 semver-3.0.2

FoamyGuy, 2024-04-01:
The `requirements.txt` file in this repo does seem weird to me. https://github.com/adafruit/circup/blob/main/requirements.txt

There are imports in there that don't seem to actually be required as far as I can tell. `Jinja2` caught my eye due to having a dependabot PR open for it. To the best of my knowledge circup doesn't use that, I tried grepping in the repo and the only reference I can find to it is in requirements.txt file. 

This file is very long I'm guessing there may be others that aren't actually necessary, this looks like it may be output from someone running `pip freeze` in a python environment that wasn't a small contained venv for this project only.

I think you are right that it could be removed @dhalbert. 

I don't know about the other questions specifically but I do think it would also be nice to switch to pyproject.toml I think for consistency with the majority of libraries.

PR #214 (2024-06-03): switch from setuppy to pyprojecttoml
Closes #189 

**Changes:**

* switch from `setup.py` to `pyproject.toml`
* clean up requirements list. there is only one `dev` extra list used for development and building.
* add `.devcontainer` for reproduceable dev environment in VSCode

**Notes**

* local build works with `python -m build`,  however I'm not sure that automatic versioning with `setuptools-scm` works correctly. 
* :boom: ~~ci script is broken (need to fix before merge). See comment further on.~~
* config files like `.pylint.rc`  could be consolidated to `pyprojecct.toml` ,  this is somethi [... cut]
closed 2024-06-03 by FoamyGuy