adafruit/circup #193: `semver` package breaking change: `isvalid()` -> `is_valid()`

aricooperdavis, 2023-11-16:
Hi circup developers!

When [the `python-semver` dependency](https://github.com/python-semver/python-semver) updated to version 3.x.x this included [the breaking change of renaming the `isvalid()` method to `is_valid()`](https://github.com/python-semver/python-semver/pull/398).

This breaks [`__init__.py` line 1517](https://github.com/adafruit/circup/blob/main/circup/__init__.py#L1517) which refers to the old method.

I'll submit a PR with the fix and reference this issue.

Many thanks for your hard work!
Ari

---

- Platform/operating system (i.e. Raspberry Pi with Raspbian operating system,
  Windows 32-bit, Windows 64-bit, Mac OSX 64-bit, etc.): 

`Fedora 39`

- Python version (run `python -version` or `python3 -version`):

`Python 3.12.0`

- Error message you are receiving, including any Python exception traces:

```python
Traceback (most recent call last):
  File "/home/ari/.local/bin/circup", line 8, in <module>
    sys.exit(main())
             ^^^^^^
  File "/usr/lib/python3.12/site-packages/click/core.py", line 1130, in __call__
    return self.main(*args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.12/site-packages/click/core.py", line 1055, in main
    rv = self.invoke(ctx)
         ^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.12/site-packages/click/core.py", line 1657, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
                           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.12/site-packages/click/core.py", line 1404, in invoke
    return ctx.invoke(self.callback, **ctx.params)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.12/site-packages/click/core.py", line 760, in invoke
    return __callback(*args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.12/site-packages/click/decorators.py", line 26, in new_func
    return f(get_current_context(), *args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/ari/.local/lib/python3.12/site-packages/circup/__init__.py", line 1517, in update
    if isinstance(module.bundle_version, str) and not VersionInfo.isvalid(
                                                      ^^^^^^^^^^^^^^^^^^^
AttributeError: type object 'Version' has no attribute 'isvalid'. Did you mean: 'is_valid'?
```

- List the steps to reproduce the problem below (if possible attach code or commands
  to run):

```cmd
$ circup --path /run/media/ari/MERLYN_FS_1 update
Found device at /run/media/ari/MERLYN_FS_1, running CircuitPython 8.2.7.
A newer version of CircuitPython (8.2.8) is available.
Get it here: https://circuitpython.org/board/feather_bluefruit_sense
Found 15 module[s] needing update.
Please indicate which modules you wish to update:

Traceback (most recent call last): [...]
```

aricooperdavis, 2023-11-16:
See https://github.com/adafruit/circup/pull/194/ for PR.

imnotjames, 2023-11-16:
I didn't see this breakage but I definitely see where this was referenced -- I must have missed that code path when testing?

Commit in particular that makes the change on the semver side - https://github.com/python-semver/python-semver/commit/5485b6bd2d03139b3bae82c7fd50195a5595322c

closed 2023-11-16 by dhalbert