adafruit/circup #195: Documentation for PyProject.toml

rgrizzell, 2023-11-19:
I'm working on creating some CircuitPython libraries, so I was very happy to see #188 get merged in. However, documentation around this new feature is missing and others may not be aware of it's recent inclusion.

FoamyGuy, 2023-11-19:
Agree'd, some documentation around this would be great! I was primarily focused on the functionality. 

@rgrizzell are you interested / willing to submit a PR that adds information to the README about the new pyproject.toml dependency functionality?

rgrizzell, 2023-11-19:
@FoamyGuy I can handle that. I just need to get fully up to speed on the PyProject format and how that ties in with library development. This is my first introduction to both, so once I get that working I'll share my notes in this issue. If everything looks good, I'll create the PR.

rgrizzell, 2024-01-09:
@FoamyGuy Can you provide me a basic how-to? I seem to be misunderstanding something basic and I can't seem to get this feature to work. None of the changes made to `pyproject.toml` seem to be picked up by Circup.

```
[circup]
circup_dependencies = ["adafruit_display_text"]
```

```
$ circup install -a
Found device at /media/rgrizzell/CIRCUITPY, running CircuitPython 9.0.0-alpha.2-105-gbbff8b5638-dirty.
Searching for dependencies for: []
```

closed 2024-02-07 by rgrizzell

FoamyGuy, 2024-02-07:
I'm going to re-open this because I still think it would be great to have documentation for this.

@rgrizzell sorry, I missed your previous message. If you're still interested in this:

Where is the pyproject.toml file that you edited?

In order for this functionality to work the section must be in pyproject.toml file of a library that is inside of a bundle, and the latest released version of that library must have these items in the pyproject.toml file. 

Then when circup installs that library it will read its toml file and install any dependencies listed there. 

As an example the Cedargrove Chime library contains the dependency list in it's pyproject.toml section here: https://github.com/CedarGroveStudios/CircuitPython_Chime/blob/main/pyproject.toml#L49-L50

and therefore when circup installs that library it picks up its dependency also:
```
❯ circup install cedargrove_chime
Found device at /media/timc/CIRCUITPY, running CircuitPython 9.0.0-beta.0.
Searching for dependencies for: ['cedargrove_chime']
Ready to install: ['cedargrove_chime', 'cedargrove_midi_tools']

Installed 'cedargrove_chime'.
Installed 'cedargrove_midi_tools'.

```