adafruit/circup #196: All lib foo.mpy files updated each use of circup with 9.0.0  alpha 5 and 6

b-blake, 2023-12-25:
Thank you for opening an issue on an Adafruit Python library repository.  To
improve the speed of resolution please review the following guidelines and
common troubleshooting steps below before creating the issue:

- **Do not use GitHub issues for troubleshooting projects and issues.**  Instead use
  the forums at http://forums.adafruit.com to ask questions and troubleshoot why
  something isn't working as expected.  In many cases the problem is a common issue
  that you will more quickly receive help from the forum community.  GitHub issues
  are meant for known defects in the code.  If you don't know if there is a defect
  in the code then start with troubleshooting on the forum first.

- **If following a tutorial or guide be sure you didn't miss a step.** Carefully
  check all of the steps and commands to run have been followed.  Consult the
  forum if you're unsure or have questions about steps in a guide/tutorial.

- **For Python/Raspberry Pi projects check these very common issues to ensure they don't apply**:

  - If you are receiving an **ImportError: No module named...** error then a
    library the code depends on is not installed.  Check the tutorial/guide or
    README to ensure you have installed the necessary libraries.  Usually the
    missing library can be installed with the `pip` tool, but check the tutorial/guide
    for the exact command.

  - **Be sure you are supplying adequate power to the board.**  Check the specs of
    your board and power in an external power supply.  In many cases just
    plugging a board into your computer is not enough to power it and other
    peripherals.

  - **Double check all soldering joints and connections.**  Flakey connections
    cause many mysterious problems.  See the [guide to excellent soldering](https://learn.adafruit.com/adafruit-guide-excellent-soldering/tools) for examples of good solder joints.

If you're sure this issue is a defect in the code and checked the steps above
please fill in the following fields to provide enough troubleshooting information.
You may delete the guideline and text above to just leave the following details:

- Platform/operating system (i.e. Raspberry Pi with Raspbian operating system,
  Windows 32-bit, Windows 64-bit, Mac OSX 64-bit, etc.):  **Windows 10**

- Python version (run `python -version` or `python3 -version`):  **circuitpython 9.0.0-alpha-5 or 6**

- Error message you are receiving, including any Python exception traces:  **No error message***

- List the steps to reproduce the problem below (if possible attach code or commands
  to run): **SEE BELOW**

*****************************************
Each time I run circup on my adafruit Trinkit RP2040 running CP 9.0.0 alpha 5 or 6, all the foo.mpy files in the lib folder get updated each time.  Installed foo.py files are not updated.  I am running circup on a Windows 10 machine.  circup is the most recent version.  Below is the batch file I run on my PC.

Bruce

`@ECHO OFF
cd C:\Users\Bruce\AppData\Local\Packages\PythonSoftwareFoundation.Python.3.10_qbz5n2kfra8p0\LocalCache\local-packages\Python310\Scripts\
dir
echo.
circup --version
echo.
:start
SET choice=
SET /p choice=Do you want to upgrade? [Y/n]: 
IF NOT '%choice%'=='' SET choice=%choice:~0,1%
IF   '%choice%'=='Y' GOTO yes
IF   '%choice%'=='y' GOTO yes
IF   '%choice%'=='N' GOTO no
IF   '%choice%'=='n' GOTO no
IF   '%choice%'==''  GOTO yes
ECHO "%choice%" is not valid
ECHO.
GOTO start

:no
rem ECHO Do all of the NO things here!
rem PAUSE
EXIT

:yes
rem ECHO Do all of the YES things here!

echo "Q: Misc"
circup --path Q: update --all
timeout 5
ECHO.
dir
timeout 30`

b-blake, 2024-02-06:
Also CP 9.0.0 beta.0

tannewt, 2024-02-07:
I bet circup doesn't know how to parse the newer mpy files for version number.

PR #198 (2024-02-13): Handle mpy version checking for Circuitpython 9
Fixes #165.
Fixes #196.

- Handle .mpy files with bytecode version `C\x06`. The `__version__` string is not findable in as straightforward a way because the way dictionary entries are in the .mpy has changed, so just look for the first string that looks like a version number.

Tested on 8.2.9 and 9.0.0-beta.0, including upgrading and downgrading between those versions.