adafruit/circup #203: CIRCUP 1.6.1 WARNINGS

b-blake, 2024-03-06:
Verson 1.6.1

I have several MCUs with all the libraries on the MCU.  When I ran version 1.6.1:

WARNINGS were issued saying typing_extensions and pyansi were not found as part of the dependencies section.  
70 +/- libraries were updated. (68 to 73)
I assume they are MicroPython libraries not CircuitPython libraries, therefore the six or so warnings.

Windows 10 machine.

b-blake, 2024-03-07:
It happened on several boards with large EPROM storage available. I had updated them about two weeks ago with CIRCUP and all was well. Boards listed below that I use, have most if not all AdaFruit libraries. I do it because I don't like to stop when adding a feature to find the library. CIRCUP 1.6.1 does not tell which libraries are triggering the 5 or 6 warnings. CP does have a few MP libraries built in. My WAG is that CIRCUP looks for the *include* lines in the library being checked and then looks to see if that library is locally available. If it is not it looks at the AF library list to fine and load it. When it can't it issues the warning. My best guess.

I am not using any MP libraries. I think the AF libraries are using MP libraries.

Arduino Nano RP2040 Connect with rp2040
Adafruit QT2040 Trinkey with rp2040
Adafruit Feather RP2040 with rp2040
Challenger RP2040 WiFi/BLE with rp2040

dhalbert, 2024-03-07:
`typing_extensions` is a CPython library we don't need: it's only used for type-checking. I don't see where `pyansi` is coming from. These are not MicroPython libraries.

But don't worry about these warnings -- they are not important. The number of updates make sense: the update logic was not working properly until 1.6.1.

b-blake, 2024-03-07:
I am using 1.6.1.

dhalbert, 2024-03-07:
Until 1.6.1, `circup update` was not working properly.

b-blake, 2024-03-07:
Dan,  Please ignore the following at your pleasure.

Put CP 8.2.10 on a Feather RP2040
Fill the fil/ directory with all libraries from MPY library bundle dated 20240301
Run circup update
15 libraries updated
3 Warnings given

With no changes to CP or the lib/ files from the above:
Run circup update a second time.
No updates, no warnings

So the three warnings are an expected output?

dhalbert, 2024-03-07:
Could you give the exact output when updating 20240301, which is slightly out of date (as I think you wished)?

b-blake, 2024-03-07:
CircUp, A CircuitPython module updater. Version 1.6.1
Found device at U:, running CircuitPython 8.2.10.
Found 15 module[s] needing update.
Updated adafruit_aws_iot
Updated adafruit_connection_manager
Updated adafruit_dash_display
Updated adafruit_gc_iot_core
Updated adafruit_json_stream
Updated adafruit_oauth2
Updated adafruit_requests
Updated adafruit_templateengine
Updated adafruit_azureiot
Updated adafruit_ble
Updated adafruit_fona
Updated adafruit_io
Updated adafruit_minimqtt
Updated adafruit_pyportal
Updated adafruit_wiznet5k
Checking 15 updated module[s] for missing requirements.
WARNING:
        typing-extensions is not a known CircuitPython library.
WARNING:
        typing-extensions is not a known CircuitPython library.
WARNING:
        pyasn1 is not a known CircuitPython library.
Ready to install: []

mikeysklar, 2024-03-07:
@dhalbert I just installed Circup-1.6.1 (2024-03-06) on MacOS using a CP 9.0.0-beta.2 drop on a Feather RP2040.

I can reproduce the warnings that @b-blake has mentioned by installing th **adafruit_gc_iot_core**:

```
sklarm@grazie Downloads % circup install adafruit_gc_iot_core
Found device at /Volumes/CIRCUITPY, running CircuitPython 9.0.0-beta.2.
Searching for dependencies for: ['adafruit_gc_iot_core']
WARNING:
        typing-extensions is not a known CircuitPython library.
WARNING:
        pyasn1 is not a known CircuitPython library.
Ready to install: ['adafruit_binascii', 'adafruit_bus_device', 'adafruit_connection_manager', 'adafruit_esp32spi', 'adafruit_gc_iot_core', 'adafruit_hashlib', 'adafruit_jwt', 'adafruit_logging', 'adafruit_minimqtt', 'adafruit_requests', 'adafruit_rsa']

Installed 'adafruit_binascii'.
Installed 'adafruit_bus_device'.
'adafruit_connection_manager' is already installed.
Installed 'adafruit_esp32spi'.
Installed 'adafruit_gc_iot_core'.
Installed 'adafruit_hashlib'.
Installed 'adafruit_jwt'.
Installed 'adafruit_logging'.
'adafruit_minimqtt' is already installed.
Installed 'adafruit_requests'.
Installed 'adafruit_rsa'.
sklarm@grazie Downloads %
```

dhalbert, 2024-03-07:
`pyasn1` is an optional import for part of the `adafruit_rsa` library. `circuitptython_typing` is only used for type checking.

So the updates and the warnings are completely expected when updating from the March 1 bundle. We could suppress warnings for those particular libraries and it would be less surprising. I will open an issue about that. I'm going to close this because it is working as expected.

dhalbert, 2024-03-07:
Reopening: I will just use this issue.

Consider suppressing warnings for these and perhaps other libraries:

- [ ] typing_extensions
- [ ] asn1
- [ ] circuitpython_typing

b-blake, 2024-03-16:
@dhalbert did you apply these changes to 1.6.2?

dhalbert, 2024-03-16:
No, this hasn't been done yet.

FoamyGuy, 2024-04-25:
This is resolved in #208, those warnings should be suppressed by default now.

closed 2024-04-25 by FoamyGuy