adafruit/circup #204: web workflow does not work with CP 8.2.x due to API differences

vladak, 2024-03-06:
Trying web workflow against ESP32 Feather V2 running CP 8.2.10:
```
$ ./venv/bin/circup --verbose --host 172.40.0.11 --password XXX list
Logging to /Users/foo/Library/Logs/circup/circup.log

03/06/2024 22:03:23 INFO: ### Started Circup ###
03/06/2024 22:03:23 INFO: Checking for a newer version of circup
03/06/2024 22:03:23 INFO: Requesting redirect information: https://github.com/adafruit/circuitpython/releases/latest
03/06/2024 22:03:23 INFO: Tag: '8.2.10'
Found device at http://:XXX@172.40.0.11, running CircuitPython 8.2.10.
03/06/2024 22:03:24 INFO: List
03/06/2024 22:03:24 INFO: Using bundles: adafruit/Adafruit_CircuitPython_Bundle, adafruit/CircuitPython_Community_Bundle, circuitpython/CircuitPython_Org_Bundle
03/06/2024 22:03:24 ERROR: list indices must be integers or slices, not str
Traceback (most recent call last):
  File "/Users/vladimirkotal/Pi/circup/venv/lib/python3.9/site-packages/circup/__init__.py", line 598, in find_modules
    device_modules = backend.get_device_versions()
  File "/Users/vladimirkotal/Pi/circup/venv/lib/python3.9/site-packages/circup/backends.py", line 608, in get_device_versions
    return self.get_modules(urljoin(self.device_location, self.LIB_DIR_PATH))
  File "/Users/vladimirkotal/Pi/circup/venv/lib/python3.9/site-packages/circup/backends.py", line 64, in get_modules
    return self._get_modules(device_url)
  File "/Users/vladimirkotal/Pi/circup/venv/lib/python3.9/site-packages/circup/backends.py", line 367, in _get_modules
    return self._get_modules_http(device_lib_path)
  File "/Users/vladimirkotal/Pi/circup/venv/lib/python3.9/site-packages/circup/backends.py", line 388, in _get_modules_http
    for entry in r.json()["files"]:
TypeError: list indices must be integers or slices, not str
There was a problem: list indices must be integers or slices, not str
```
Checking the traffic dump the JSON returned for the request is merely an array.

It works with CP 9.0.0 beta2.

vladak, 2024-03-07:
I wonder if the web workflow API is documented somewhere.

anecdata, 2024-03-07:
https://docs.circuitpython.org/en/latest/docs/workflows.html#web

vladak, 2024-03-07:
The difference is that CP 8.2.10 has `web_api_version` = 2 and returns `[]` on empty directory listing, while CP 9.0.0 has `web_api_version` = 4 and returns dictionary/object with the `files` value being empty array.

I think the answer to this is to check the API version and refuse anything strictly lower than 4.

commit eb108fe5 (2024-03-09): refuse any device with web API version stricly less than 4

fixes #204
closed 2024-03-09 by dhalbert