adafruit/circup #206: Add support for v2 web workflow

dhalbert, 2024-03-09:
There is support for v4 web workflow but not v2. See #205.

FoamyGuy, 2024-03-11:
I started working on this in a branch here: https://github.com/FoamyGuy/circup/tree/webworkflow_api_2_support

However I was unsuccessful at testing and completing the development of it. I'm running into persistent issues with the webworkflow reporting that the storage is readonly even though my boot.py file has mounted it as writeable and I have confirmed that REPL is able to write file successfully and USB PC connection is not able to write files. 

This issue seems to be describing the exact thing I ran into: https://github.com/adafruit/circuitpython/issues/8986

I've tested on a Feather S3 TFT and Feather S2 TFT with 8.2.10 and 8.1.0 and seen the same issue on both. 

There is a fix for that issue in 9.x. But that also contains the change to webworkflow api v4 which is what circup now targets.

Unless that fix will be backported to 8.x it means that even if we add support in circup for v2 some portion of devices will still not be able to use circup successfully due to that issue making webworkflow believe device is readonly.

dhalbert, 2024-03-11:
GIven the imminence of CircuitPython 9.0.0, I think we can skip doing this.

closed 2024-03-11 by dhalbert