adafruit/circup #213: Python 3.12 problems

dhalbert, 2024-05-01:
I installed circup 1.8.0 in a venv with Python 3.12.3 on Ubuntu 24.04. Two problems arose:

1. circup expects `setuptools` to be installed because it uses `pkg_resources`. `pkg_resources` is deprecated. From https://setuptools.pypa.io/en/latest/pkg_resources.html:
> Use of pkg_resources is deprecated in favor of [importlib.resources](https://docs.python.org/3.11/library/importlib.resources.html#module-importlib.resources), [importlib.metadata](https://docs.python.org/3.11/library/importlib.metadata.html#module-importlib.metadata) and their backports ([importlib_resources](https://pypi.org/project/importlib_resources), [importlib_metadata](https://pypi.org/project/importlib_metadata)). Some useful APIs are also provided by [packaging](https://pypi.org/project/packaging) (e.g. requirements and version parsing). Users should refrain from new usage of pkg_resources and should work to port to importlib-based solutions.

2. After installing `setuptools` to work around problem 1, I got this when trying a simple install:
```sh
$ circup install neopixel
Found device at /media/halbert/CIRCUITPY, running CircuitPython 9.0.4.
Searching for dependencies for: ['neopixel']
Traceback (most recent call last):
  File "/home/halbert/.py/bin/circup", line 8, in <module>
    sys.exit(main())
             ^^^^^^
  File "/home/halbert/.py/lib/python3.12/site-packages/click/core.py", line 1157, in __call__
    return self.main(*args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/halbert/.py/lib/python3.12/site-packages/click/core.py", line 1078, in main
    rv = self.invoke(ctx)
         ^^^^^^^^^^^^^^^^
  File "/home/halbert/.py/lib/python3.12/site-packages/click/core.py", line 1688, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
                           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/halbert/.py/lib/python3.12/site-packages/click/core.py", line 1434, in invoke
    return ctx.invoke(self.callback, **ctx.params)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/halbert/.py/lib/python3.12/site-packages/click/core.py", line 783, in invoke
    return __callback(*args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/halbert/.py/lib/python3.12/site-packages/click/decorators.py", line 33, in new_func
    return f(get_current_context(), *args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/halbert/.py/lib/python3.12/site-packages/circup/commands.py", line 347, in install
    device_modules = ctx.obj["backend"].get_device_versions()
                     ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/halbert/.py/lib/python3.12/site-packages/circup/backends.py", line 73, in get_device_versions
    return self.get_modules(os.path.join(self.device_location, self.LIB_DIR_PATH))
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/halbert/.py/lib/python3.12/site-packages/circup/backends.py", line 63, in get_modules
    return self._get_modules(device_url)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/halbert/.py/lib/python3.12/site-packages/circup/backends.py", line 772, in _get_modules
    return _get_modules_file(device_lib_path, self.logger)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/halbert/.py/lib/python3.12/site-packages/circup/shared.py", line 87, in _get_modules_file
    metadata = extract_metadata(source, logger)
               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/halbert/.py/lib/python3.12/site-packages/circup/shared.py", line 169, in extract_metadata
    result["__version__"] = version.decode("utf-8")
                            ^^^^^^^^^^^^^^^^^^^^^^^
UnicodeDecodeError: 'utf-8' codec can't decode byte 0x80 in position 2: invalid start byte
```

sjev, 2024-05-05:
I've picked up #189  . This one is related.  However, I  did not encounter issuse in a devcontainer running Python 3.12.3. 

`pip install .` worked without issues. Please provide info on how to reproduce.

dhalbert, 2024-05-05:
@sjev
Here is a reproduction on Ubuntu 24.04, running in a venv named `.py`. `setuptools` is installed in the venv, and then I uninstall it:
```sh
(.py) halbert@cod:~$ circup list
Found device at /media/halbert/CIRCUITPY, running CircuitPython 9.1.0-beta.1-16-g8734076e04-dirty.
All modules found on the device are up to date.
(.py) halbert@cod:~$ pip list |grep setuptools
setuptools                               69.5.1
(.py) halbert@cod:~$ pip3 uninstall setuptools
Found existing installation: setuptools 69.5.1
Uninstalling setuptools-69.5.1:
  Would remove:
    /home/halbert/.py/lib/python3.12/site-packages/_distutils_hack/*
    /home/halbert/.py/lib/python3.12/site-packages/distutils-precedence.pth
    /home/halbert/.py/lib/python3.12/site-packages/pkg_resources/*
    /home/halbert/.py/lib/python3.12/site-packages/setuptools-69.5.1.dist-info/*
    /home/halbert/.py/lib/python3.12/site-packages/setuptools/*
Proceed (Y/n)? y
  Successfully uninstalled setuptools-69.5.1
(.py) halbert@cod:~$ circup list
Traceback (most recent call last):
  File "/home/halbert/.py/bin/circup", line 5, in <module>
    from circup import main
  File "/home/halbert/.py/lib/python3.12/site-packages/circup/__init__.py", line 9, in <module>
    from circup.shared import DATA_DIR, BAD_FILE_FORMAT, extract_metadata, _get_modules_file
  File "/home/halbert/.py/lib/python3.12/site-packages/circup/shared.py", line 14, in <module>
    import pkg_resources
ModuleNotFoundError: No module named 'pkg_resources'
```
If you can't reproduce this make sure `setuptools` has not been installed via apt. E.g. try `import setuptools` in Python.

sjev, 2024-05-06:
@dhalbert  Thank you, this is clear. Now I see that I misunderstood the original issue.