adafruit/circup #215: Pre-commit broken on Python 3.12.3

sjev, 2024-05-06:
- Platform/operating system:  **docker, python:3-slim** 

- Python version (run `python -version` or `python3 -version`):  **3.12.2**

- Problem:  `pre-commit run -all-files` crashes. (log below)

- How to reproduce:

1. checkout `main`
2. open it in docker container:

`docker run -it --rm -v "$(pwd)":/workspace -w /workspace python:latest bash`

3. run these:

```
git config --global --add safe.directory /workspace
pip install pre-commit
pre-commit run --all-files
```

**error log (relevant part)**

```
 File "/tmp/pip-build-env-l7g3nkvd/overlay/lib/python3.12/site-packages/pkg_resources/__init__.py", line 2191, in <module>
              register_finder(pkgutil.ImpImporter, find_on_path)
                              ^^^^^^^^^^^^^^^^^^^
          AttributeError: module 'pkgutil' has no attribute 'ImpImporter'. Did you mean: 'zipimporter'?
          [end of output]
```

icfly2, 2024-06-12:
You need to run this first:

https://github.com/open-mmlab/mim/issues/242#issuecomment-2125079539

sjev, 2026-09-21:
closing as stale

closed 2026-09-21 by sjev