adafruit/circup #227: Make circup frozen library aware

mMerlin, 2024-06-08:
Feature request.

When using circup to install a library, it also installs all of the dependencies. That is usually the correct thing to do, but when those dependencies are frozen, it wastes space.
suggestion: add a flag, or make it the default, to exclude frozen dependencies during installs. They should be shown as already installed.

Variation. If installing with --py, frozen library dependencies might be desired.
suggestion: add a flag to install frozen dependencies when --py is used.
suggestion: when installing source (--py) for frozen libraries have a flag to override the target folder. Putting them in the root folder instead of lib would let them be found before the frozen version.
Putting them in a user specified folder would allow access control by manipulating sys.path. Something like '/debug', '/unfrozen', or '/thawed'.

Some extra documentation (circup --help) about interaction with frozen libraries could be a good idea.

- List the steps to reproduce the problem below (if possible attach code or commands
  to run): **LIST REPRO STEPS BELOW**

On discord in help-with-circuitpython
https://discord.com/channels/327254708534116352/537365702651150357/1248796673115816078

This arose when using
```bash
circup install --py adafruit_pyportal
```
then trying to add debug prints to the imported dependence libraries. Some worked, some didn't because they were frozen, but there was no indication that was the problem.

justmobilize, 2024-06-08:
This could install the stubs, and then pull frozen libs from there...

mMerlin, 2024-06-08:
Stubs are part of blinka ?
I don't think stubs are enough depending which scenario is in play. Not to be able to add debug code to the actual library running in the circuitpython mcu.

The issue is that the actual source for the frozen libs "are" being installed, but that code is never referenced because frozen is before lib in sys.path. Either they should not be installed at all (because the frozen reference will be used), or they needed to be installed somewhere before frozen in sys.path. If the mpy files are being installed, that "should" have identical functionality to what is in frozen. .py files can be modified, changing the functionality.

justmobilize, 2024-06-08:
The stubs list the actual frozen libs. So for circup, it can find the matching stub, parse for the list and find if it's in there.

It has the same information you find on the board page (they both get it from the same place).

From that point it can give the opinion of installing in root or not at all.

In some cases installing them could cause the board to crash since it takes more memory to load them.