adafruit/circup #232: Circup 2.0.4 not working after update from 1,9

dsssssssss9, 2024-08-04:
-:  Ubuntu 20.04.6

- Python version (run `python -version` or `python3 -version`):  3.12.4

- Error message you are receiving, including any Python exception traces:  

$ circup
Traceback (most recent call last):
  File "/home/jkd/.local/bin/circup", line 5, in <module>
    from circup import main
  File "/home/jkd/.local/lib/python3.12/site-packages/circup/__init__.py", line 9, in <module>
    from circup.shared import DATA_DIR, BAD_FILE_FORMAT, extract_metadata, _get_modules_file
  File "/home/jkd/.local/lib/python3.12/site-packages/circup/shared.py", line 15, in <module>
    import requests
  File "/usr/lib/python3/dist-packages/requests/__init__.py", line 43, in <module>
    import urllib3
  File "/usr/lib/python3/dist-packages/urllib3/__init__.py", line 7, in <module>
    from .connectionpool import HTTPConnectionPool, HTTPSConnectionPool, connection_from_url
  File "/usr/lib/python3/dist-packages/urllib3/connectionpool.py", line 11, in <module>
    from .exceptions import (
  File "/usr/lib/python3/dist-packages/urllib3/exceptions.py", line 2, in <module>
    from six.moves.http_client import IncompleteRead as httplib_IncompleteRead
ModuleNotFoundError: No module named 'six.moves'

- List the steps to reproduce the problem below (if possible attach code or commands
  to run): **LIST REPRO STEPS BELOW**

Type Circup in console

dhalbert, 2024-08-04:
If you type `which -a python3` what does it say?

How did you install Python 3.12?

It looks like you have a mixed environment going on right now: some packages are in `~/.local/lib/python3.12`, but some are in `/usr/lib/python3/dist-packages`, which are probably compatible with the `/usr/bin/python3` version of Python, which would not be 3.12.

The best thing to do to use Python 3.12 is to use venv or virtualenv to install a consistent environment for 3.12. You _may_ be able to adapt the instructions in this guide: https://learn.adafruit.com/python-virtual-environment-usage-on-raspberry-pi

Have you considered updating from Ubuntu 20.04 to 24.04?

dsssssssss9, 2024-08-04:
> If you type `which -a python3` what does it say?
*$ which -a python3
/usr/bin/python3
** 

> How did you install Python 3.12?

https://www.howtogeek.com/install-latest-python-version-on-ubuntu/

I only did this because circup would NOT upgrade past v1.9 & was giving errors that hinted to me that i needed later version of Python to perform the upgrade

The best thing to do to use Python 3.12 is to use venv or virtualenv to install a consistent environment for 3.12. You _may_ be able to adapt the instructions in this guide: https://learn.adafruit.com/python-virtual-environment-usage-on-raspberry-pi

Will give that a go & report back

> Have you considered updating from Ubuntu 20.04 to 24.04?
I will need to upgrade my SSD before doing  any upgrade as storage om current SSD is getting full . 

I may even install a fresh copy of 24.04 on it & then try to replicate my current setup as best i can -  probably about time as this linux has been upgraded through  many versions over the years. I suspect that may be part of the problem ?

dhalbert, 2024-08-04:
It is not a good idea to install the latest version of Python as `/usr/bin/python3`. That may well break system utilities and the like that depend on the version of Python installed as `/usr/bin/python3`. It is fine to use deadsnakes to installed Python 3.12 as `python3.12` and invoke it as that (and not try to reset the default version to python3.12).

> I may even install a fresh copy of 24.04 on it & then try to replicate my current setup as best i can - probably about time as this linux has been upgraded through many versions over the years. I suspect that may be part of the problem ?

Yes, this is what I'd recommend. If you are fine with installing a fresh 24.04, that is very much the best choice, especially since your 20.04 version may now have version skews that will make it unstable.

dhalbert, 2024-08-25:
> This will hopefully means that the problems i was having will not reoccur?

I think so.

You can use `pipx`, or if you are installing many things with `pip` or `pipx`, you can create a "personal" venv and just use it all the time. I've created one, in the `~/.py` directory, and have this in my `.bash_aliases`:
```sh
source ~/.py/bin/activate
```
So the `.py` venv gets activated on every new terminal windows that's opened.