adafruit/circup #237: ImportError: no module named 'adafruit_register'

RnoldR, 2025-01-07:
- Platform/operating system (i.e. Raspberry Pi with Raspbian operating system,
  Windows 32-bit, Windows 64-bit, Mac OSX 64-bit, etc.):  
  - Pimoroni Pico Plus 2W
  - Host: Ubuntu 24.04

- Python version (run `python -version` or `python3 -version`):  3.12.3

- Error message you are receiving, including any Python exception traces:  ImportError: no module named 'adafruit_register'

- List the steps to reproduce the problem below (if possible attach code or commands
  to run): 

import sys
import math
import time
import board
import busio
import digitalio
import adafruit_vl53l1x
import adafruit_bno055

From Thonny 4.0.1. I receive the following error message:

>>> %Run -c $EDITOR_CONTENT
Traceback (most recent call last):
  File "<stdin>", line 8, in <module>
  File "adafruit_bno055.py", line 36, in <module>
ImportError: no module named 'adafruit_register'

The adafruit_vl53l1x imports succesfully, the _bno055 does not. The same  code runs correctly on a Raspberry Pico 2W. I know it is about a missing library but I think this is about a dependency in the bno055 library. The image I used to initialize the Pico Plus 2W is: adafruit-circuitpython-pimoroni_pico_plus2w-en_US-20250107-main-PR9940-26283dc.uf2. The library bundle used is adafruit-circuitpython-bundle-9.x-mpy-20241221.zip.

RnoldR, 2025-01-07:
I forgot to mention this problem seems to be related to #90

Neradoc, 2025-01-07:
Hi, what is the circup command you are running ?
I get `adafruit_register` when I install `adafruit_bno055`:
```
❯ circup  install adafruit_bno055
Found device at /Volumes/CIRCUITPY, running CircuitPython 9.2.1.
Searching for dependencies for: ['adafruit_bno055']
Ready to install: ['adafruit_bno055', 'adafruit_bus_device', 'adafruit_register']

Installed 'adafruit_bno055'.
Installed 'adafruit_bus_device'.
Installed 'adafruit_register'.
```
It's in there: https://github.com/adafruit/Adafruit_CircuitPython_BNO055/blob/main/requirements.txt

Neradoc, 2025-01-07:
This is the circup repository though.
You are looking for support, please go [to the forums](https://forums.adafruit.com/) or [the discord server](https://adafru.it/discord).

In this case you just need to copy the missing `adafruit_register` library from the bundle to your board.
[You can read the guide for some advice on that](https://learn.adafruit.com/welcome-to-circuitpython/circuitpython-libraries#example-importerror-due-to-missing-library-2978246).
You can also [follow the guide to use circup](https://learn.adafruit.com/keep-your-circuitpython-libraries-on-devices-up-to-date-with-circup) to install the dependencies for you.

closed 2025-01-07 by Neradoc