adafruit/circup #249: CircUp updates updated modules.

b-blake, 2025-04-28:
I keep getting these libraries being updated even though they were updated 2 minutes before.

`Found 2 module[s] needing update.
WARNING: Library adafruit_rgb_display repo has incorrect __version__
        metadata. Circup will assume it needs updating.
        Please file an issue in the library repo.
        https://github.com/adafruit/Adafruit_CircuitPython_RGB_Display.git
Updated adafruit_rgb_display
WARNING: Library ansi_escape_code repo has incorrect __version__
        metadata. Circup will assume it needs updating.
        Please file an issue in the library repo.
        https://github.com/s-light/CircuitPython_ansi_escape_code.git
Updated ansi_escape_code
Checking 2 updated module[s] for missing requirements.`

b-blake, 2025-04-28:
Issue filed with libraries also.

FoamyGuy, 2025-05-21:
Closing this now, the RGB display library has a release which fixes this, ansi escape codes is a community library and has an issue created for this. The author of that library will have to make the release for it.

closed 2025-05-21 by FoamyGuy